remrc/Self-Balancing-Cube #3: The motor is not moving

pockethmoe, 2022-06-15:
I have connected the circuit board, and the gyroscope test is normal, there is a detection sound when starting, but the motor does not move, the replacement motor failure is still the same, what is the reason?
cube.ino  
pinMode(DIR1, OUTPUT);
  ledcSetup(PWM1_CH, BASE_FREQ, TIMER_BIT);
  ledcAttachPin(PWM1, PWM1_CH);
  Motor1_control(0);

Motor1_control(0);？？？

pockethmoe, 2022-06-15:
There is a strange problem in debugging today：aruiido IDE The serial port monitoring gyroscope data is similar to your value，I have modified the data according to the video and update to ESP32，But when trying to balance，But when trying to balance，In case of edge test, the same is true, but if it is not at the balance point position, the motor will not rotate，Could you provide me with an email? I'll send you the video. Please help me，thankyou。

pockethmoe, 2022-06-15:
https://user-images.githubusercontent.com/107524980/173880648-efb92002-d51a-4b31-8a54-2a4a9a9a6e40.mp4

pockethmoe, 2022-06-15:
I don't know what the problem is. The situation is the same after replacing the motor

pockethmoe, 2022-06-15:
measure brake in esp32 d26 spot  have 3.5V，The wiring is also checked according to the circuit diagram and no problem is found

remrc, 2022-06-15:
Only full speed? You didn’t mix PWM and BRAKE? Please lift the cube on the edge slowly. And show me video.

pockethmoe, 2022-06-16:
I connected according to the circuit diagram and checked the corresponding signal points of the motor. The physical wiring is normal。ESP32  32、25、18pin is PWM，26pin is brake and shared by three motors。

pockethmoe, 2022-06-16:
It's very strange. Today, the test is carried out and the function is realized!!! I don't quite understand what caused the fault before. But thank you very much for sharing the excellent project. Thank you! I'll test it later and ask again if I have any questions.

pockethmoe, 2022-06-16:
https://user-images.githubusercontent.com/107524980/173984145-3e71f89c-23d0-4cb1-a4f6-a19b679e3803.mp4

pockethmoe, 2022-06-16:
https://user-images.githubusercontent.com/107524980/173984174-c3ac30c7-d807-4d79-ae0c-58043d8e822f.mp4

remrc, 2022-06-16:
Oh, yes! I am very glad that you have achieved success.

pockethmoe, 2022-06-16:
I have another problem, adjusting settings K1 K2 K3 via Bluetooth，What is the function of this parameter，If it is modified, is it written to esp32 or reset after shutdown and restart?

pockethmoe, 2022-06-16:
Although the function has been realized from the verification, there are still many questions about the code to communicate with you. Although I have realized the function, I still want to learn more from you. Thank you for your support and help.

remrc, 2022-06-16:
K1 K2 K3 - controllers parameters. This is not need but you can change it. Yes, it resets after restart. Change it only if you know what you are doing.

pockethmoe, 2022-06-17:
Thank you very much for trying to see

closed 2022-06-18 by remrc