remrc/Self-Balancing-Cube #4: Not able to balance on all sides

freeifier, 2022-06-28:
Hey, I just wanted to ask if a heavier battery might be causing the balancing problems. On one site it works fine, on another it tries but seems to slow to react.

My build differs in that I am using a bigger battery maybe 2 to 3 times heavier and also a bit larger so the center of mass is higher up and also have not achieved the 70g per reaction wheel.

It worked fine on the one side but maybe it's not enough.

If you have time it would be great if you would help.
Otherwise I'll figure it out eventually and try increasing the weight first.

freeifier, 2022-07-01:
I figured it out myself, for people building this project you have to be careful when switching around motor connections, this will also impact how the cube reacts to specific axis. The best is to physically switch the motor connectors if the wrong motor spins for the given axis. Otherwise you will have to change the values which the motors will receive.

remrc, 2022-07-01:
A bigger battery is not good, but I didn't understand what you meant: "have to be careful when switching around motor connections, this will also impact how the cube reacts to specific axis"? You got the axles mixed up? Off course, this is important. But this is easy understand and swap if necessary.

freeifier, 2022-07-01:
My Motor connections differ from yours in that, that the 
1->3, 2->1 and 3->2. So I changed the pin output in the code, but than also the directions was mirrored, wo I changed that in Code too. Then I got two sides Wirkung first one but removed the battery than the second worked too.

Then the 3rd side die not work at all, I Bad to change the value given to Motor1_Control from pwm_Y to -pwm_X.

Then all sides worked fine but edge point, if had to change the values too so after I understood how the cube is counteracting the forces I tested 2 Motors controlling one Y axis and then figured out X axis and then it worked. 
So awesome.

remrc, 2022-07-01:
Oh, not needed to change pins in code, not needed to change motors directions. Only swap motor connections. I not declare what motor is 1, what 2 and what 3. So it is not clear which motor is on which side of the cube. But it is easy to determine. Only swap motor connections!

freeifier, 2022-07-01:
Yeah, but the cables wouldn't fit otherwise, so I took this path the nice thing is that I understand the code and know where everything is, learned also about PID controllers on the way so nice excersise.
It is also nice code solid structure found the stuff quite well.

remrc, 2022-07-01:
Yes, understanding what is where - good. But there is no PID here...

closed 2022-07-01 by remrc

freeifier, 2022-07-01:
Ok it seemed that way not that familiär with pid bit you have a proportional term than one which responds to Motorspeed and one Which you can tune with I thought it might stand for Intergral.

What type of controller is it then ?