remrc/Self-Balancing-Cube #11: Schematics

OpenSourceDistilling, 2024-08-22:
I think the transistor is wired correctly in the Nano schematic and incorrectly on the other one. 

I believe it should be an N-type transistor, wired to the middle pin with the 6.8K ohms resistor to D27 instead of to ground.

[image]

remrc, 2024-09-02:
I didn't write anywhere in the ESP32 schematic that it's a 2N2222. I use different transistor. But if you're using a 2N2222, yes, the Nano schematic is correct.
I drew it like mine and I didn't think there would be such a problem... It's strange... Connecting the buzzer is the simplest thing, how to do it can be found on the Internet in 1 minute.

malebuffy, 2024-10-31:
> I think the transistor is wired correctly in the Nano schematic and incorrectly on the other one.
>
> I believe it should be an N-type transistor, wired to the middle pin with the 6.8K ohms resistor to [...]

Would you have the fritzing file you could share with me?